COAST-Lab/Open-Water-Level #2: Break up PUBLISH_STATE to do separate tasks in separate states

SUPScientist, 2023-09-14:
PUBLISH_STATE currently collects and averages data, prints to screen, saves to micro-SD card, and, if cell is activated, publishes to cloud. This should be separated into several discrete states. Or, if it's unnecessary to have state machine structure (since only a single sequence is ever used), at least move code chunks out into separate functions. E.g., a data collection/averaging function, a data printing function, a data saving function, and a data publishing function.

SUPScientist, 2023-10-10:
Hi, Hailey, I'm replying to parts of your email via GitHub so we can track it over here in the context of this issue. Here are a few comments/next steps on what I see from your recent work.

Instead of dropping the file you've been working on into the main repo's top directory, edit the code of the .ino file directly. In this case, that would be https://github.com/COAST-Lab/Open-Water-Level/blob/main/Firmware/SLR_Boron_Maxbotix_MB7092_cm/src/SLR_Boron_Maxbotix_MB7092_cm.ino. You can just copy and paste your new code into that file, save, commit, etc. and then git/GitHub will show what are called "diffs" (or differences) from the old version to your modifications. Then we'll be able to track the changes you're making as you make and commit them. Nice work creating a new branch, starting some changes, and committing them, and connecting things to this issue.

SUPScientist, 2023-10-10:
Assigning this to @hseppey since she's taken ownership of this issue.

hseppey, 2023-10-16:
When committing the changes I have made so far, should I commit them to the main branch or to the new branch I made for this issue?

SUPScientist, 2023-10-17:
If changes relate to this issue, then commit those changes to the branch you created for this issue. That way the main branch will remain "as is" so people can use the old code until we can verify the functionality of your new code.