COAST-Lab/Open-Water-Level #3: Add wiring instructions to getting started readme

SUPScientist, 2023-09-28:
The [Getting Started Readme](https://github.com/COAST-Lab/Open-Water-Level/tree/main/Firmware#getting-started-with-cell-enabled-water-level-sensing) in the firmware subdirectory has a section on using the water level sensor after two intro exercises ((1) Blink LED and (2) write data to SD card). 

This [section on water level sensing/logging](https://github.com/COAST-Lab/Open-Water-Level/tree/main/Firmware#practice-water-level-sensor-full-code-boron-and-adalogger) has good software/firmware guidance but no hardware/wiring instructions. 

Add wiring instructions to this readme file so that others can learn how to set up the hardware. Note that there's a related example here: https://github.com/COAST-Lab/Open-Water-Level/blob/main/Firmware/Sensor%20How-To.md#connecting-the-sensor-wiring. However, that related example is for a slightly different sensor. It should still work for the new and improved sensor, perhaps with minor modifications.

BentleySettin, 2023-10-02:
In the assembly section of the Readme it states that the water level sensor uses OLED FeatherWing, RTC FeatherWing, Maxbotix, and Feather microcontroller. I used a Boron, RTC featherwing, and Maxbotix. Do you want me to put my wiring instructions even though I used different electronics?

SUPScientist, 2023-10-02:
This issue is specifically related to the [Getting Started Readme](https://github.com/COAST-Lab/Open-Water-Level/tree/main/Firmware#getting-started-with-cell-enabled-water-level-sensing) as linked in the initial post. That doesn't state anything about the Feather microcontroller. The main readme in the repo's top directory definitely needs updating but this issue is specific to the [Getting Started Readme](https://github.com/COAST-Lab/Open-Water-Level/tree/main/Firmware#getting-started-with-cell-enabled-water-level-sensing), so don't worry about other readme files in dealing with this issue.

BentleySettin, 2023-10-02:
Gotcha, I was looking at the wrong Readme. I just updated the first line in the Boron and Adalogger section and the first line in the Practice full code section. Do you want me to add images? I see that the example you sent previously included images also.

SUPScientist, 2023-10-02:
Excellent. Nice work. I'm looking at https://github.com/COAST-Lab/Open-Water-Level/commit/a9faafad45e96351497a7c57d1692e0f236c3dd9 and it looks like exactly the sorts of updates we needed. Yes, one good image showing the wire connections would help, and if you think there's anything else needed for someone to learn how to complete the physical setup following that guide. Thanks.

SUPScientist, 2023-11-28:
@BentleySettin, I think you may have completed this work but maybe not yet posted instructions and an image of the wire connections to this repository. Could you post the image in a relevant subdirectory and link to it from here, or if it's already in the repo somewhere, post a link here and I'll close this issue?

BentleySettin, 2023-11-28:
Here are the instructions: https://github.com/TinyCamML/Boron-and-OpenMV-Basics/commit/7c6f8948191d8e4c47ddbd254060045436634b3d

The wire connections are in this document and here it is in the "Boron-and-OpenMV-Basics" repo: https://github.com/TinyCamML/Boron-and-OpenMV-Basics/commit/7c6f8948191d8e4c47ddbd254060045436634b3d

SUPScientist, 2023-11-28:
I meant for the water level sensor project in this repo (not the OpenMV one). This is an old thread that I'm revisiting because we never closed it when you were working on it two months ago.

BentleySettin, 2023-11-30:
Here is the document with added instructions and linked images within document if this is what you are referring to : 

https://github.com/COAST-Lab/Open-Water-Level/commit/8dd33a6dfeefda5d930ce61ae7488d19acd91512

SUPScientist, 2023-11-30:
Looks good, thanks! Closing as complete...

closed 2023-11-30 by SUPScientist