COAST-Lab/Open-Water-Level #4: Beautify your code!

SUPScientist, 2023-10-10:
Arduino/Wiring/Particle's Device OS/C/C++/etc... do not care about if/how you indent your code, but it is far more readable if nested loops (loops inside loops, or if/for blocks, etc.) show increasing levels of indentation. This isn't hard to do by hand (keyboard) and VS Code tries to help you out along the way. But it can occasionally be very helpful to use a free online formatter/beautifier to do it for you.

Check out https://codebeautify.org/cpp-formatter-beautifier as an example. Put in your code chunk, or entire script, and see it spit out nicely formatted code. Note that the link above is for C++ (which covers Arduino/Particle) but there are other flavors for Python and other languages, too. 

@BentleySettin, @hseppey, et al., please use that occasionally to ensure nicely formatted and more readable code. It won't just help others; it will help you follow your own code better!

SUPScientist, 2025-05-06:
This issue should be converted to general info on the COAST Lab wiki or other getting started guide rather than an always-open issue in this repo.

BentleySettin, 2025-05-07:
Sounds good. I can move this to "other notes" in the COAST Lab repo unless you prefer another place.

SUPScientist, 2025-05-07:
Rachel is working on a wiki to eventually live [here](https://github.com/COAST-Lab/COAST-Lab.github.io/wiki) for stuff like this, so we can sit on it for a sec.