COAST-Lab/Open-Water-Level #14: Enclosure subdirectory is badly out of date

SUPScientist, 2023-10-19:
Old SolidWorks drawings for laser cut enclosure haven't been used in 3 years. Update with new housing drawings when possible.

SUPScientist, 2023-10-26:
Updated parts files (designed in Fusion, exported to .stl for 3D printing) are now here: https://github.com/COAST-Lab/Open-Water-Level/tree/main/Enclosure/PVC-End-Caps. Several modifications still needed:

- [ ] Threads didn't show up in STL. That is, they were in F3D file, but didn't export. This support article seems to know about the issue but didn't solve it: https://www.autodesk.com/support/technical/article/caas/sfdcarticles/sfdcarticles/Threads-do-not-show-up-in-an-exported-STL-file-from-Fusion-360.html.
- [ ] The pipe gripper that seals the top of the top end cap needs a lip to sit on. It currently rests on an inner slope of the top end cap's ugly angled flange and doesn't rest down on a lip like it should, and therefore doesn't seal the way it should. Make a lip (1.9–2" diameter).
- [ ] Better flange needed for solar panel with screw holes for mounting panel.
- [ ] Hole needed for solar power cable.

SUPScientist, 2023-11-06:
See https://github.com/COAST-Lab/Open-Water-Level/issues/15 for issue specific to top end cap

SUPScientist, 2023-11-16:
@acl3053, check out a few updates here: https://github.com/COAST-Lab/Open-Water-Level/tree/main/Enclosure/PVC-End-Caps. Note that GitHub renders STLs nicely in the browser! 

I modified your bottom end cap design to open up the part of the hole that is not threaded. I used an NPT tap/drill chart to choose the new diameter. 

(Side note: when I tried to export your end cap design into the Inventor file format (.ipt; in case in the future someone wanted to open it in Inventor), Fusion 360 created a folder that contains also the threaded part in addition to the part you designed so that I guess it can do the subtraction.)

acl3053, 2023-11-16:
@SUPScientist, has the thread itself changed at all? I thought that the subtraction method didn't seem to work based on the print I made a couple of weeks ago. I made the part again in Inventor to see if that would work but threads do not model in Inventor when exported to .stl. I uploaded it to my fork here: https://github.com/acl3053/Open-Water-Level/blob/main/Enclosure/PVC-End-Caps/WaterLevel_BottomCap_InventorTest.ipt if anyone can figure out a way around this limitation.

SUPScientist, 2023-11-16:
No, the thread itself has not changed. The small change I made as mentioned above might allow an NPT tap to open up the existing threads slightly, though. It's not ideal to both print threads and tap threads (usually I pick one or the other!), but this may be a workable solution. We'll see.

acl3053, 2023-11-17:
Here is link for a solution to the lack of thread modeling in Fusion 360 (using the above stated subtraction method) for future reference: https://forums.autodesk.com/t5/fusion-360-design-validate/modeled-npt-tapped-threads/td-p/11653018.

SUPScientist, 2024-01-04:
Improvements may be necessary, but closing with https://github.com/COAST-Lab/Open-Water-Level/commit/fcdc1154d399747b5f7f4f890c0da2b4f97ea5d8 given marked improvements to design and file organization

closed 2024-01-04 by SUPScientist