COAST-Lab/Open-Water-Level #18: Particle update

Stephen-Lail, 2024-01-24:
In the current update of Particle, the SYSTEM_MODE(SEMI_AUTOMATIC) did not work and the Boron 404X was still searching for signal (green light was flashing). It can be fixed by replacing it with SYSTEM_MODE(MANUAL) and having #include "Particle.h" at the top of the code. There may be other solutions to this issue but this is the one I found to work.

SUPScientist, 2024-02-19:
Likely related to the updates described here: https://community.particle.io/t/new-particle-project-template/65629. Note that adding the `#include` statement is due to the shift from using a .ino file as default to .cpp. `#include` was invisible "under the hood" when we used to use .ino. Not sure about `SEMI_AUTOMATIC` yet but investigating this further...

SUPScientist, 2024-11-06:
Issue not with code and has been resolved. Closing/no commit needed.

closed 2024-11-06 by SUPScientist