COAST-Lab/Open-Water-Level #23: Add table of contents to top of main readme

SUPScientist, 2024-02-19:
A key improvement to the repository's main readme (i.e., the readme.md file that renders when you go to this repository's highest level folder) would be to add a table of contents that links to the important documents that can be found elsewhere. E.g., the getting started guide, the fabrication guide, ... what else would someone want to know when they first showed up at this repo?

SUPScientist, 2024-03-04:
@dylan9081, I just pulled and merged changes from https://github.com/COAST-Lab/Open-Water-Level/pull/26. 

The repo's main readme should be the next target. See comment above. I recommend starting with a blank page rather than working off of the existing readme. Write an outline of the information that you think should be described on the repo's homepage (or in other words the main readme file). Share that outline with me (e.g., online word doc in onedrive) so that I can contribute, and then start to fill it out using markdown. We can pull in text from the existing readme, but the existing readme is so irrelevant at this stage that I'd prefer you start with a blank page.

The main readme is meant to address the question: "what would you want to know if you found yourself staring at this repository's homepage/main readme?" The answer can come in the form of both well organized text as well as hyperlinks.

SUPScientist, 2024-03-22:
Some improvements in https://github.com/COAST-Lab/Open-Water-Level/pull/27. Still in progress.

SUPScientist, 2024-03-25:
Noting more progress in https://github.com/COAST-Lab/Open-Water-Level/pull/29 and several recent commits such as https://github.com/COAST-Lab/Open-Water-Level/commit/52d44031ac312f37193f55b04293dc6768f483ba. @dylan9081, please give this another read and click on the links on the main readme to confirm everything works. You may close this issue if you think the readme accomplishes its goal, or continue adding to it if not (let me know if the latter via a comment here).

dylan9081, 2024-03-26:
I did a couple of small text/grammar edits, but everything looks good. All links worked.

closed 2024-03-26 by dylan9081