COAST-Lab/Open-Water-Level #32: Fit parts inside in more protected and reliable way

SUPScientist, 2024-04-09:
In new polycase enclosure, parts are essentially stuffed inside: 
[image]
Find a better way to mount parts inside with, for example, a 3D-printed chassis that holds the electronics and wires in specified locations.

SUPScientist, 2024-04-09:
@dylan9081, do you want to take a stab at this one?

SUPScientist, 2024-04-10:
See also https://github.com/COAST-Lab/Open-Water-Level/blob/main/Fabrication/Datasheets/polycase_2d_drawings.pdf for interior dimensions

gracelower, 2024-07-18:
Chassis designs:
[Fabrication/PVC-Enclosure/Enclosure-and-Assembly](https://github.com/COAST-Lab/Open-Water-Level/tree/791c02cdef5b158903b8f5a916721b1455ee3ed0/Fabrication/PVC-Enclosure/Enclosure-and-Assembly)
[Fabrication/Polycase-Enclosure/Enclosure-and-Assembly](https://github.com/COAST-Lab/Open-Water-Level/tree/791c02cdef5b158903b8f5a916721b1455ee3ed0/Fabrication/Polycase-Enclosure/Enclosure-and-Assembly)

closed 2024-11-06 by SUPScientist