COAST-Lab/Open-Water-Level #53: Adapting for deeper range and use within a well

gotjoshua, 2024-11-06:
I am considering to drive into this diy for our 7.5m deep well.

I see that the brand of sensor you use also has some with deeper range:
https://maxbotix.com/products/mb7066

Can you give an indication how complicated it will be too adapt the code and design to work with a different sensor? 

Not asking you to make that happen, just for a reality check and some guidance. If I go for it, I'll be happy to submit a PR

SUPScientist, 2025-03-05:
Given that we're using the sensor's analog output and that your proposed sensor adaptation uses that as well, code changes would likely be straightforward. I imagine it would simply be a matter of proper ADC count to distance conversion, which for us happens in https://github.com/COAST-Lab/Open-Water-Level/blob/831b5fd010759d1795c42bbcbe54730bb942a039/Firmware/SLR_Boron_Maxbotix_MB7092_cm/src/SLR_Boron_Maxbotix_MB7092_cm.cpp#L134.

(PS: apologies for the terribly belated reply. We hadn't properly configured notifications for this repo in our organization and am seeing this issue for the first time just now.)

gotjoshua, 2025-03-06:
Many thanks for your reply... We are also very slow to get started with the project, but it's great to know that you are around if we run into issues!