COAST-Lab/Open-Water-Level #66: Look into SDI-12 interface for particle products

BentleySettin, 2025-03-20:
CTD meter uses SDI-12 Interface and I need to determine how to integrate this with Particle Boron.

SUPScientist, 2025-03-20:
Thanks for creating this issue for the sake of project management, @BentleySettin! Please document your process and knowledge here as well both for posterity and so that I can chime in. I sent some ideas via email last week, so you can copy those into this thread. It sounded from Alex as though the UART block on the Boron could be used, albeit with a different library and not the typical `Serial1` that we use all the time.

SUPScientist, 2025-05-05:
Picked up over here (https://github.com/BentleySettin/SDI12_Boron/issues/1) so closing in this repo.

closed 2025-05-05 by SUPScientist