COAST-Lab/Open-Water-Level #69: Investigate logic to test Boron with max publish state at 11 minutes.

BentleySettin, 2025-05-07:
Is there a way to force the Boron to undergo an ISMI switch or full hardware reset during data collection to determine effectiveness of an 11 minute max publishing time?

BentleySettin, 2025-07-28:
@SUPScientist Sounds great, I will flash this today and let it run for 2-3 days to see if there are any hiccups.

SUPScientist, 2025-07-29:
Noting that this issue is closely related to, if not identical to, #68.

BentleySettin, 2025-07-31:
Sounds good, I closed it with a thread back to this conversation.

@SUPScientist 
I have had the [new code ](https://github.com/COAST-Lab/Open-Water-Level/issues/69#issuecomment-3127709863) running for the past two days and am seeing some **percieved** connectivity issues within the lab. One of two things usually happen. 1)The first time it wakes up or the few times I have reset it, I have seen some new behavior. The boron will connect to the particle cloud and then for about 2-5 minutes go through a cycle of flashing cyan, two orange, some green and so on. According to particle files this means that it could not connect to internet. After these few minutes the boron flashes slow cyan (connected to internet) then white and then off. 2) The boron wakes up and uses the full time to try to connect to internet but doesn't. 

I majority of the runs the boron will not wake up on the hour. I am testing code we know works to determine if this is an internet or firmware issue. I have switched the antenna too but that did not change anything.

BentleySettin, 2025-07-31:
On the particle console it has notes for waterLevel_0008 : "This Boron was deployed with the waterLevel_0005 device at BMC-RAND from 10-Mar-2025 through 25-Mar-2025 but stopped transmitting on 17-Mar-2025. The BMC-RAND site received a new Boron after testing isolated this Boron as seemingly the problem. New Boron installed at 1415 EDT 25-Mar-2025." This may be the issue as this was a boron that was not transmitting well at FMB, granted we also concluded that whole site may have been off with poor internet reception.

BentleySettin, 2025-08-01:
After switching to a new Boron (waterLevel_0010) the updated code is successfully running. The previous boron is in tinker mode with an antenna trying to connect to the cloud. After about 2 hours, still no success.

SUPScientist, 2025-08-01:
OK, let's separate these two issues even though they became apparent through the same process. Testing the timeout_test branch to solve the title issue here is ongoing (through the weekend to get multiple 24 hr cycles). 

Separately, we need a new issue for troubleshooting (or, if needed, disposing of) waterLevel_0008.

closed 2026-02-13 by SUPScientist