COAST-Lab/Open-Water-Level #78: Create a separate eventname for testing water level sensors

BentleySettin, 2025-06-30:
Currently all published data from water level sensors go to "waterLevel" eventname which is connected to our google sheet. It would be helpful for during testing scenerios to use a separate eventname so that data from tests and real deployments are not intermingled.

SUPScientist, 2025-07-25:
That eventname also pushes data to the CORMP webpages, so for the same reason, it makes sense to have another eventname as well as an additional google sheet.

### Recommendations

- [ ] 1. Add a line below https://github.com/COAST-Lab/Open-Water-Level/blob/1709e3a53e8c88ff1828647cdf860f8a6e215f0f/Firmware/OWL_Boron_Maxbotix_MB7092_cm/src/OWL_Boron_Maxbotix_MB7092_cm.cpp#L63 that is something like `const char * eventName = "testing";` Leave one uncommented and comment out the other.
- [ ] 2. Add `Log.info()` statements where data are printed out in loop below that also prints out the `eventName` so that we have it right in front of our eyes.
- [ ] 3. Add note in pre-deployment checklist to confirm that `PUBLISHING` is set to 1 and `eventName` to waterLevel.
- [ ] 4. Create new google sheet that catches the testing events.

BentleySettin, 2025-07-30:
**Firmware:**

I am currently using this chunk of code as my eventname:
" const char * eventName = "waterLevel_0008"; "

**Google Sheets Integration & Webhook Creation:**

I used protocol from our repo: https://github.com/COAST-Lab/Open-Water-Level/tree/main/Particle_Integration
There is currently a webhook for waterLevel and waterLevel_0008.

**Current Status:**

As of now waterLevel_0008 seems to be having connectivity issues within the lab and is therefore not reporting anything to the either google sheet or particle console. When I forced a reset, data printed on our master google sheet but not the separate WL_0008 google sheet I created. I am investigating where I may have gone wrong or where Particle may have had some integration/webhook updates.

BentleySettin, 2025-08-01:
After integrating a new boron I am succesfully seeing water level data for the new boron (waterLevel_0010) printing on both the large sheet and the personal sensor sheet.  

I followed protocol from this page: https://github.com/COAST-Lab/Open-Water-Level/tree/main/Particle_Integration
The only change I made was that I commented out lines 50-52 from the .js.

closed 2025-08-01 by BentleySettin