COAST-Lab/Open-Water-Level #85: Three OWLs simultaneously lost cell service

SUPScientist, 2025-08-06:
### Problem
On 15-Jul-2025 at 0400, all four operational OWLs successfully sampled. Beginning at 0500, only the "RAND" site (waterLevel_0005) sampled and all three others ceased sampling for several days. 
[image]

### Hypothesized cause
Only waterLevel_0005 had been updated to Device OS 6.2.1 (due to issues at the Randall site that we have been troubleshooting). All three other deployed units were still running 4.2.0. This seems likely to be more than a coincidence. 

Suggest updating all other units (KERR, 23ST, and CHES) to Device OS 6.2.1 and latest FW.

SUPScientist, 2025-08-06:
Tagging @BentleySettin and @efarq just as an FYI based on yesterday's conversation.

BentleySettin, 2025-08-12:
Interesting that the last time we heard from waterLevel_0003 transmission took about 7 minutes. Could the SIM have undergone an IMSI switch which somhow affected its failure? Trying to use clues from connectivity conversation with [Rikkas](https://community.particle.io/t/confusing-boron-connectivity-issues/69438/3).

SUPScientist, 2025-08-15:
Updating after several days to say that both waterLevel_0003 and waterLevel_0006 which were upgraded to Device OS 6.2.1 and latest OWL user firmware have been unresponsive for at least several days. waterLevel_0003 worked @ 100% uptime for 72 hours post update while waterLevel_0006 worked for 16 hours post update. Both dropped to 0% uptime after that elapsed time. 

Potentially of note, both these devices had Device OS upgraded via the Workbench Particle: Flash Device OS and User Firmware command (rather than Device Restore USB web tool, which is what had been used to update waterLevel_0005, which is still going very strong (> 99% uptime since 24-June while running Device OS 6.2.1). 

Recommend updating to 6.3.3. We should also examine options for a system reset using hibernate mode according to https://community.particle.io/t/difficulty-reconnecting-to-cell-network/69817/7.

SUPScientist, 2025-10-06:
Re: 

> Recommend updating to 6.3.3. We should also examine options for a system reset using hibernate mode according to https://community.particle.io/t/difficulty-reconnecting-to-cell-network/69817/7. 

Hibernate is not likely an option for the Boron 404X according to https://docs.particle.io/reference/device-os/api/sleep-sleep/hibernate-systemsleepmode/. The Boron could likely go into Hibernate mode, but it cannot wake from a timer—only a WKP pin. A WKP pin is not a good solution in our case.

However, the other useful tidbit from that same Particle community thread is rickkas's: 

> You should wait 12-15 minutes of failing to connect to cellular, not 5 minutes, to allow Device OS to do a full modem reset after 10 minutes

We need to try `MAX_TIME_TO_PUBLISH_MS = 900000`.