COAST-Lab/Open-Water-Level #90: WL_0006 blinking SOS

BentleySettin, 2025-09-22:
When WL_0006 was retrieved 9/18/2025 it was flashing the associasted SOS signal. Once taken into the lab to be flashed with the updated firmware and code, it was left over the weekend to run. When I returned the following Monday the boron was flashing red again after the time left in the tinker state and updated firmware/code.  To troubleshoot, I intially reflashed the code and watched the sensor during the morning. The sensor succesfully reported at 9am but then at 10am ran for almost 30 minutes (even with a coded time out). I have plugged the sensor into my computer now to monitor any weird activity that may show up on the serial monitor. I am seeing some interesting errors on my serial monitor. 

[image]

[image]

This may or may not be related.

SUPScientist, 2025-09-25:
Oh dear. OK, thanks for trying to troubleshoot. I appreciate any and all help but you can set this aside for now to focus on your other work; I don't want to let this become too much of a distraction.

SUPScientist, 2025-10-06:
Reviving this discussion not to pull you too deep back into it but just to quickly ask if you changed anything with wl_06, @BentleySettin. Reason I'm asking is that it appears to have been working with only a single drop since 9 pm last Wednesday evening. Would be great to know if you changed/reset anything!

BentleySettin, 2025-10-07:
I have not done anything! Not sure if someone may have accidently unplugged the usb and then plugged it back in, but to my knowledge, no changes.